sstsimulator/sst-elements #2151: Bad code in pymerlin-base

kvoronin-intel, 2023-06-06:
Hi!

This line
https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/merlin/pymerlin-base.py#L276
refers to undeclared "value" (copypasted piece of code from setattr?) which leads to
Traceback (most recent call last):
...
  **File "pymerlin-base.py", line 276, in __getattr__
NameError: name 'value' is not defined. Did you mean: 'False'?**
FATAL: SSTPythonModel: Execution of model construction function failed.

Triggered by the following high-level print() attempt:
<somewhere in config.py>
    networkif.input_buf_size = "32kB"
    networkif.output_buf_size = "32kB"
-->print("networkif buf sizes = ", networkif.input_buf_size, networkif.output_buf_size)

I am not very familiar with internals of SST but this does look like a bug.

feldergast, 2023-06-21:
Thanks.  I'll take a look and put in a patch.

hughes-c, 2023-08-14:
@feldergast Did you patch this?