sstsimulator/sst-elements #2155: Application hangs for Ember Tricount motif

AKKamath, 2023-06-12:
When running the [Ember Tricount motif](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/ember/mpi/motifs/embertricount.cc) the application ends up hanging.
The specific configuration I used is attached: [tricount.txt](https://github.com/sstsimulator/sst-elements/files/11727963/tricount.txt)

I setup the tricount input file with the following parameters:
embertricount_setup 0 18 5 45 15 15 input/vertices_18.txt

From what I can see, the nodes get stuck in a circular dependency, where one node is waiting for data from another node and vice-versa. In the motif, the MPI calls are correctly non-blocking, so I'm not exactly sure why this is triggering.

SST-Elements master branch, commit 6bcddc7db8b831c7887e9794c8042e06f7d3645c (currently the latest commit)
SST-Core master branch, commit 70f56eb5d7e489e13c733e524596acce7887fd36 (currently the latest commit)

jpkenny, 2023-06-12:
Thanks for the report. The dependencies are pretty complicated.
I will attempt to reproduce the issue using your parameters and see if I can find the cause.

jpkenny, 2023-06-13:
@AKKamath I'm pretty sure that the issue here is that the graph generator is creating unconnected vertices with these parameters, but I haven't actually been able to reproduce a hang. Can you share the full input that causes the hang?

I'm still trying to understand what's going on with the graph generation.

AKKamath, 2023-06-13:
Sure. Here's my input file: [vertices_18.txt](https://github.com/sstsimulator/sst-elements/files/11730787/vertices_18.txt)

AKKamath, 2023-06-13:
That's here: [tricount.txt](https://github.com/sstsimulator/sst-elements/files/11738140/tricount.txt)
Github does not allow ".py" files, so I have sent it as a ".txt".

Running `sst tricount.py` should be enough to run this.

jpkenny, 2023-06-14:
Thanks for the input. I can reproduce the hang and I've done some initial debugging and learned a couple things:

- The deadlock only occurs with larger node counts (e.g. 64 nodes runs but 96 nodes deadlocks)
- It looks like there are ember motif instances that never execute again after enqueing a DATA_REQUEST using enQ_irecv(). Those nodes can't subsequently enqueue a waitany and can't fulfill any more data requests, so the simulation deadlocks.

I haven't found any problem in the motif logic so far. I'm concerned that there's a bug somewhere in the ember/hermes/firefly stack. A bug there could be quite subtle and take some work to find.

I believe that my initial concern about disconnected vertexes was unfounded -- I don't currently believe that to be an issue.

Debugging continues...

jpkenny, 2023-06-27:
I've uncovered a Firefly issue that is involved in this behavior:
https://github.com/sstsimulator/sst-elements/issues/2162

Fixes to follow shortly.

jpkenny, 2023-06-29:
@AKKamath the fix for this issue is merged into the devel branch. I can run this input/configuration without issue now. Let us know if you have further issues.

closed 2023-06-29 by jpkenny

AKKamath, 2023-06-29:
Thanks so much! I was trying to debug the issue for a quite a while.