sstsimulator/sst-elements #2156: memHierarchy/CacheController - clockTick - event is accepted but packet is missed.

ideal8901, 2023-06-20:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

I am building a system component that can not reveal.

In memHierarchy/CacheController - clockTick function, event is accepted but the component connected to l2 cache does not get it iin the event buffer. So the connected component loops in its own clockTick again and again while CacheController does not has the request in the buffer (It erased).

The system works for certain amount of queries, with my custom random generator with Miranda CPU. 
But at the exact same address (Since my generator is not real random), it stops because of the packet missed. 
Also the address is accessed ahead and it worked before. 

2 - Describe how to reproduce
The component is that I cannot reveal.

3 - What Operating system(s) and versions 
Linux cluster 4.15.0-162-generic #170-Ubuntu SMP Mon Oct 18 11:38:05 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

gvoskuilen, 2023-06-27:
@ideal8901 Can you provide some more detail:
- What is your input configuration, particularly for the caches?
- Have you tried compiling with debug enabled and viewing the output to see what the cache does with the request? To do this, configure sst-core with '--enable-debug' and set "debug=1; debug_level=6" in the cache parameters.

ideal8901, 2023-07-04:
The component (Component_0) has set like below. 

return new MemEventInitCoherence(getName(), Endpoint::Component_0, true /* inclusive */, true /* sends WBAck */, false /* expects WBAck */, lineSize_, true /* tracks block presence */);

They are connected.
CPU - L1 - L2 - Component_0 - DirectoryController - memcontroller

I am using simple custom random address generator. 
The debug results are below. 
As you can see, at the last two lines, their request are missed in L2 cache and never trigger Component_0.
Could you let me know possible cases please? Thank you! 

Issue request: address=0x18c3a800, length=8, operation=READ, cache line offset=0
E: 2226552              6236                 l1_cache_0           Event:Recv    (<79,0> GetS Src: mirandaCPU_0 Dst: l1_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 8 Prf: F)
E: 2234049              6257                 l2_cache_0           Event:Recv    (<79,0> GetS Src: l1_cache_0 Dst: l2_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2270163              12718                Component_0            Event:New     (<79,0> GetS Src: l2_cache_0 Dst: Component_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2270163              12718                Component_0            Event:New     (<79,0> GetS Src: l2_cache_0 Dst: Component_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2270163              12718                Component_0            Event:New     (<79,0> GetS Src: Component_0 Dst: Component_0_mem_ctrl_2 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2270820              3633                 Component_0_mem_ctrl_2 Event:New     (<79,0> GetS Src: Component_0 Dst: Component_0_mem_ctrl_2 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2301875              3683                 Component_0_mem_ctrl_2 Event:Resp    (<79,0> GetXResp Src: Component_0_mem_ctrl_2 Dst: Component_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: T VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2302293              12898                Component_0            Event:New     (<79,0> GetXResp Src: Component_0_mem_ctrl_2 Dst: Component_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: T VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2302293              12898                Component_0            Event:New     (<79,0> GetXResp Src: Component_0_mem_ctrl_2 Dst: Component_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2302293              12898                Component_0            Event:New     (<79,0> GetXResp Src: Component_0 Dst: l2_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2302950              6450                 l2_cache_0           Event:Recv    (<79,0> GetXResp Src: Component_0 Dst: l2_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
E: 2306163              6459                 l1_cache_0           Event:Recv    (<79,0> GetXResp Src: l2_cache_0 Dst: l1_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: T VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)
Issue request: address=0x18c3a800, length=8, operation=WRITE, cache line offset=0
E: 2314017              6481                 l1_cache_0           Event:Recv    (<80,0> Write Src: mirandaCPU_0 Dst: l1_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: T VA: 0x18c3a800 IP: 0x0 Size: 8 Prf: F)
E: 8404080              23540                l2_cache_0           Event:Recv    (<214,0> PutM Src: l1_cache_0 Dst: l2_cache_0 Rq: l1_cache_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: T VA: 0x0 IP: 0x0 Size: 64 Prf: F)
E: 25386627             142222               Component_0            Event:New     (<706,0> PutM Src: l2_cache_0 Dst: Component_0 Rq: l2_cache_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: T VA: 0x0 IP: 0x0 Size: 64 Prf: F)
E: 25386627             142222               Component_0            Event:New     (<706,0> PutM Src: l2_cache_0 Dst: Component_0 Rq: l2_cache_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x0 IP: 0x0 Size: 64 Prf: F)
E: 25386627             142222               Component_0            Event:New     (<706,0> PutM Src: Component_0 Dst: Component_0_mem_ctrl_2 Rq: l2_cache_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x0 IP: 0x0 Size: 64 Prf: F)
E: 25387284             40619                Component_0_mem_ctrl_2 Event:New     (<706,0> PutM Src: Component_0 Dst: Component_0_mem_ctrl_2 Rq: l2_cache_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x0 IP: 0x0 Size: 64 Prf: F)
Jupark generate: addr 0x18c3a800
Issue request: address=0x18c3a800, length=8, operation=READ, cache line offset=0
E: 154639491            433163               l1_cache_0           Event:Recv    (<5801,0> GetS Src: mirandaCPU_0 Dst: l1_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 8 Prf: F)
E: 154646988            433184               l2_cache_0           Event:Recv    (<5801,0> GetS Src: l1_cache_0 Dst: l2_cache_0 Rq: mirandaCPU_0Tid: 0 Flags: [] Addr: 0x18c3a800 (G) Data: F VA: 0x18c3a800 IP: 0x0 Size: 64 Prf: F)

ideal8901, 2023-07-07:
src/sst/elements/memHierarchy/coherencemgr/MESI_Inclusive.cc : processCacheMiss it returns stall.

However, it only loops clockTick() in cacheController without resending the request. 
That is why it never sends message to Component_0.

ideal8901, 2023-07-18:
Close this ticket.

closed 2023-07-18 by ideal8901