sstsimulator/sst-elements #2160: run .py with balar

chaojibendan, 2023-06-27:
When I try to run `sst testBalar-testcpu.py --model-options='-c gpu-v100-mem.cfg -v -x vectorAdd/vectorAdd -t cuda_calls.trace'`,  I also encountered the same issue. [SST-GPGPUSim Integration #2090](https://github.com/sstsimulator/sst-elements/issues/2090)
“sst: symbol lookup error: /wafer/wyy/software/sst-elements-library-13.0.0/build/lib/sst-elements-library/libbalar.so: undefined symbol: SST_gpgpusim_numcores_equal_check”
I followed the instructions in https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/balar/README.md to build Balar.  I don't know how to run 'setup_environment' script correctly. It seems like that there is a problem in `source setup_environment sst` .
Before that, I need to explain the modifications I made for building balar.
GCC：11.3.0
CUDA：11.8
SST：13.0.0
When I install GPGPU-Sim,
Replace line 47 in `setup_environment` with `CC_VERSION=11.3.0` to fix the issue with regex not correctly recognizing CC_VERSION，
Replace line 46 in `version_detection.mk` with `CC_VERSION := 11.3.0` and Comment out line 49.
Then I run `source setup_environment sst`.
Then I add the following line below line 170 in the Makefile:
if [ ! -f $(SIM_LIB_DIR)/libcudart_mod.so ]; then ln -s libcudart.so $(SIM_LIB_DIR)/libcudart_mod.so; fi
This is because of an issue with the check in Balar.
In the src/cuda-sim/Makefile file, I make modifications based on the instructions provided in the following link: (https://github.com/accel-sim/gpgpu-sim_distribution/pull/42/files.)
This modification addresses the compatibility issue with newer versions of Bison, such as Bison 3.8.2, which may cause compilation errors.

hughes-c, 2023-06-27:
`undefined symbol: SST_gpgpusim_numcores_equal_check” `happens if the gpgpusim environment is not properly initialized or if balar is not built correctly. During configure, did it say that balar was being built? If configure is successful, you should see:
```
cuda : YES 
balar : YES
```

chaojibendan, 2023-06-28:
yes， I'm sure.
Configuration Information (Found Dependencies):

                                 boost : No
                                  chdl : No
                                  cuda : YES
                               dramsim : No
                              dramsim3 : No
                                 dumpi : No
                                 fdsim : No
                                  glpk : No
                         goblin_hmcsim : No
                              gpgpusim : YES
                            hbmdramsim : No
                             hybridsim : No
                                  libz : YES
                                  llvm : No
                                 metis : No
                             nvdimmsim : No
                                   otf : No
                                  otf2 : No
                               pintool : No
                     ptrace_set_tracer : YES
                                  qsim : No
                             ramulator : No
                                   shm : YES
                              spinlock : No
                                 stake : No

-------------------------------------------------------
Configuration Information (Make will build the following elements):

                                 ariel : YES
                                 balar : YES
                           cacheTracer : YES
                               cassini : YES
                               CramSim : YES
                                 ember : YES
                               firefly : YES
                                hermes : YES
                              kingsley : YES
                          memHierarchy : YES
                               mercury : YES
                                merlin : YES
                               Messier : YES
                               miranda : YES
                                   mmu : YES
                              prospero : YES
                               rdmaNic : YES
                                 Samba : YES
                                shogun : YES
                  simpleElementExample : YES
                             thornhill : YES
                               vanadis : YES
                             VaultSimC : YES
                                zodiac : YES

-------------------------------------------------------

hughes-c, 2023-06-28:
I can't replicate your problem with setup environment. The symbol error is usually caused by not being able to find the gpgpu-sim libraries (problem with the environment).

chaojibendan, 2023-06-29:
The problem might be here?
```
source setup_environment sst
echo $LD_LIBRARY_PATH
```
The output is `/wafer/wyy/software/gpgpu-sim_distribution/lib/gcc-11.3.0/cuda-11080/sst:`
But there is no such directory, only /wafer/wyy/software/gpgpu-sim_distribution/lib/gcc-11.3.0/cuda-11080/release.

chaojibendan, 2023-07-01:
Maybe it would be best for me to show my complete compilation process.
ubuntu 22.04
gcc 11.3.0
cuda  11.8
First install sst-core.
```
cd ~/software
wget https://github.com/sstsimulator/sst-core/releases/download/v13.0.0_Final/sstcore-13.0.0.tar.gz
tar xfz sstcore-13.0.0.tar.gz
cd sstcore-13.0.0/
export SST_CORE_HOME=~/software/sstcore-13.0.0/build
export SST_CORE_ROOT=~/software/sstcore-13.0.0
./configure --prefix=$SST_CORE_HOME --disable-mpi --disable-mem-pools
make all
make install
export PATH=$SST_CORE_HOME/bin:$PATH
```
Then  install gpgpusim.
```
cd ~/software
git clone https://github.com/gpgpu-sim/gpgpu-sim_distribution.git
cd gpgpu-sim_distribution
export CUDA_INSTALL_PATH=/usr/local/cuda
export GPGPUSIM_ROOT=$HOME/software/gpgpu-sim_distribution
# Replace line 47 in setup_environment with CC_VERSION=11.3.0
# Replace line 46 in version_detection.mk with CC_VERSION := 11.3.0  and Delete line 49
# Add the following line below line 170 in the Makefile:
# if [ ! -f $(SIM_LIB_DIR)/libcudart_mod.so ]; then ln -s libcudart.so $(SIM_LIB_DIR)/libcudart_mod.so; fi
# Replace line 132 in src/cuda-sim/Makefile with the following line：
# 	cat $(OUTPUT_DIR)/ptx.tab.h | grep "=" | sed 's/^[ ]\+//' | sed -E 's/\s+\/\*.+\*\///' | sed 's/[=,]//g' | sed 's/\([_A-Z1-9]\+\)[ ]\+\([0-9]\+\)/\1 \1/' | sed 's/^/DEF(/' | sed 's/ /,"/' | sed 's/$$/")/' | sed '/YYerror/d;/YYEOF/d;/YYEMPTY/d;/YYUNDEF/d;'> $(OUTPUT_DIR)/ptx_parser_decode.def
# Replace line 134 in src/cuda-sim/Makefile with the following line：
#	cat $(OUTPUT_DIR)/ptx.tab.h | grep "=" | sed -E 's/^ +//' | sed -E 's/\s+\/\*.+\*\///' | sed 's/[=,]//g' | sed -E 's/([_A-Z1-9]+).*/\1 \1/' | sed 's/^/DEF(/' | sed 's/ /,"/' | sed 's/$$/")/' | sed '/YYerror/d;/YYEOF/d;/YYEMPTY/d;/YYUNDEF/d;' > $(OUTPUT_DIR)/ptx_parser_decode.def
source setup_environment
make
```
Then install sst-elements
```
cd ~/software
wget https://github.com/sstsimulator/sst-elements/releases/download/v13.0.0_Final/sstelements-13.0.0.tar.gz
tar xfz sstelements-13.0.0.tar.gz
cd sst-elements-library-13.0.0/
export SST_ELEMENTS_HOME=$HOME/software/sst-elements-library-13.0.0/build
export SST_ELEMENTS_ROOT=$HOME/software/sst-elements-library-13.0.0
./configure --prefix=$SST_ELEMENTS_HOME --with-sst-core=$SST_CORE_HOME --with-cuda=$CUDA_INSTALL_PATH --with-gpgpusim=$GPGPUSIM_ROOT
# cuda:yes balar:yes
make all
make install
```
usage
```
cd $SST_ELEMENTS_ROOT/src/sst/elements/balar
cd tests/
make -C vectorAdd
LD_PRELOAD=/wafer/wyy/software/accel-sim-framework/util/tracer_nvbit/others/cuda_api_tracer_tool/cuda_api_tracer/cuda_api_tracer.so  ./vectorAdd/vectorAdd
```
Then I got the first error:
```
------------- NVBit (NVidia Binary Instrumentation Tool v1.5.5) Loaded --------------
NVBit core environment variables (mostly for nvbit-devs):
            NVDISASM = nvdisasm - override default nvdisasm found in PATH
            NOBANNER = 0 - if set, does not print this banner
---------------------------------------------------------------------------------
         INSTR_BEGIN = 0 - Beginning of the instruction interval where to apply instrumentation
           INSTR_END = 4294967295 - End of the instruction interval where to apply instrumentation
      START_GRID_NUM = 0 - Beginning of the kernel gird launch interval where to apply instrumentation
        END_GRID_NUM = 4294967295 - End of the kernel launch interval where to apply instrumentation
    COUNT_WARP_LEVEL = 1 - Count warp level or thread level instructions
    EXCLUDE_PRED_OFF = 0 - Exclude predicated off instruction from count
   ACTIVE_FROM_START = 1 - Start instruction counting from start or wait for cuProfilerStart and cuProfilerStop
       MANGLED_NAMES = 1 - Print kernel names mangled or not
        TOOL_VERBOSE = 0 - Enable verbosity inside the tool
----------------------------------------------------------------------------------------------------
cudaGetExportTable: UUID = 0x6e 0x16 0x3f 0xbe 0xb9 0x58 0x44 0x4d 0x83 0x5c 0xe1 0x82 0xaf 0xf1 0x99 0x1e 
cudaGetExportTable: UUID = 0x35 0x77 0xf 0x1b 0x9 0x2e 0x3 0x48 0xa4 0x8e 0x5 0x6f 0xc4 0x23 0x96 0x8d 
cudaGetExportTable: UUID = 0xbf 0xdb 0x43 0x2d 0xbf 0x3c 0x5a 0x4a 0x94 0x5e 0xb3 0x40 0x29 0xe8 0x1e 0x75 
cudaGetExportTable: UUID = 0x21 0x31 0x8c 0x60 0x97 0x14 0x32 0x48 0x8c 0xa6 0x41 0xff 0x73 0x24 0xc8 0xf2 
cudaGetExportTable: UUID = 0x42 0xd8 0x5a 0x81 0x23 0xf6 0xcb 0x47 0x82 0x98 0xf6 0xe7 0x8a 0x3a 0xec 0xdc 
cudaGetExportTable: UUID = 0xb1 0x5 0x41 0xe1 0xf7 0xc7 0xc7 0x4a 0x9f 0x64 0xf2 0x23 0xbe 0x99 0xf1 0xe2 
cudaGetExportTable: UUID = 0xa6 0xb1 0xff 0x99 0xec 0xc4 0xc9 0x4f 0x92 0xf9 0x19 0x28 0x66 0x3d 0x55 0x85 
cudaGetExportTable: UUID = 0xf8 0x8c 0xc9 0x3e 0x53 0xfd 0x9e 0x46 0xba 0x59 0x1e 0x2b 0x87 0x3e 0xf 0x91 
Segmentation fault (core dumped)
```
But it also generates  `cuda_calls.trace`
So I ran:  `sst testBalar-testcpu.py --model-options='-c gpu-v100-mem.cfg -v -x vectorAdd/vectorAdd -t cuda_calls.trace'`
Then I got the second error:
```
Completed configuring the cuda-test model
BalarTestCPU[testcpu/balarTestCPU.cc:76:BalarTestCPU] Trace file: cuda_calls.trace, cuda_executable: vectorAdd/vectorAdd
BalarTestCPU[testcpu/balarTestCPU.cc:162:createGPUReqFromPacket] creating GPU request cpu, CUDA Function enum GPU_REG_FAT_BINARY; packet address: 0
sst: symbol lookup error: /wafer/wyy/software/sst-elements-library-13.0.0/build/lib/sst-elements-library/libbalar.so: undefined symbol: SST_gpgpusim_numcores_equal_check
```
I'm really confused with these.

chaojibendan, 2023-07-02:
Maybe I know where the problem is. SST 13.0.0 still uses `sst-gpgpusim`，not `gpgpusim`. So when I use `gpgpusim`， I can build balar. But I can't trace cuda api because `cuda_calls.trace`  is pointing to sst-gpgpusim instead of gpgpusim. 
Now gpgpusim supports CUDA 11, while sst-gpgpusim only supports CUDA 10. Additionally, there are certain issues with compiling bison and gcc at higher versions, as I made modifications during my initial compilation. Perhaps these are problems that can be addressed in the future.

hughes-c, 2023-07-03:
@William-An  Ah. Yeah, sst-gpgpusim is supposed to be merged with gpusim at some point. I will check on that status.

hughes-c, 2025-05-28:
New balar integration release in SST-15.

closed 2025-05-28 by hughes-c