sstsimulator/sst-elements #2170: [Merlin] pymerlin failure for multiple ranks per node

jpkenny, 2023-07-06:
pymerlin will throw an index error with the following minimal reproducer configuration using multiple ranks per node.

```
import sst
from sst.merlin.base import *
from sst.merlin.endpoint import *
from sst.merlin.interface import *
from sst.merlin.topology import *

from sst.ember import *

if __name__ == "__main__":

    PlatformDefinition.setCurrentPlatform("firefly-defaults")

    topo = topoSingle()
    topo.link_latency = "20ns"
    topo.num_ports = 2

    router = hr_router()
    router.flit_size = "8B"
    router.xbar_bw = "50GB/s"
    router.link_bw = "12GB/s"
    router.input_buf_size = "16kB"
    router.output_buf_size = "16kB"
    topo.router = router

    networkif = ReorderLinkControl()
    networkif.link_bw = "12GB/s"

    ep = EmberMPIJob(0,4)
    ep.network_interface = networkif
    ep.addMotif("Init")
    ep.addMotif("Fini")

    system = System()
    system.setTopology(topo,2)
    system.allocateNodes(ep,"linear")

    system.build()
```

Input will run successfully if topo.num_ports is set to 4 (number of nodes matches number of Ember ranks).

jpkenny, 2023-07-06:
These two changes (replacing commented out lines) will allow the configuration to succeed but simulations hang.

```
    def _topology_config_callback(self, variable_name, value):
        if not value: return
        self._lockVariable(variable_name)
                
        # Set the variables
        self._num_nodes = value.getNumNodes()
        #self._endpoints = [None for i in range(self._num_nodes)]
        self._endpoints = [None for i in range(self._num_nodes * self.allocation_block_size)]
        
        if self._areVariablesLocked(["topology","allocation_block_size"]):
            #self._available_nodes = [i for i in range(self._num_nodes // self.allocation_block_size)]
            self._available_nodes = [i for i in range(self._num_nodes)]
```

feldergast, 2023-07-18:
The simple reproduces is not a valid configuration.  You are asking for a 4 node job with only a 2 endpoint topology.  It should error out, though we should likely figure out how to provide a more useful error message.

jpkenny, 2023-07-18:
Possibly I just don't understand the input. Doesn't the "2" here
```system.setTopology(topo,2)```
indicate 2 ranks per node?

feldergast, 2023-07-18:
I wondered if that was the issue.  The 2 in setTopology is the block allocation size.  It's used in conjunction with multiple NICs per node, as the NICs are currently required to be on consecutive endpoint IDs.  So, if you want 2 NICs per node, you need to set the allocation size to 2.  If 4 NICs per node, set it to 4, etc.  This will ensure that endpoints are allocated in blocks of that size, ensuring that the NICs are connected to consecutive IDs.

Multiple ranks per node is set in the EmberMPIJob (different jobs can use different ranks per node).  It is the third parameter to the EmberMPIJob constructor:

EmberMPINode(job_id, num_nodes, numCores = 1, nicsPerNode = 1)

So, a 4 rank job on two nodes would be:

EmberMPINode(0, 2, 2) // two nodes, two ranks per node

In this case, you don't need to have a block allocation size of 2, but it would still be functionally correct.

jpkenny, 2023-07-18:
Ok, thanks. I probably was hunting around in inputs and found the wrong parameter (but got kinda close). I suppose "allocation_block_size" didn't really make sense.

Closing issue.