sstsimulator/sst-elements #2173: sstprospero-0-0.trace file unavailable for prospero

debtanu09, 2023-07-14:
sstprospero-0-0.trace file unavailable for prospero array tests.
Can someone help me get it, if not then just a sample file so that I can create a trace file in similar format

gvoskuilen, 2023-07-14:
The prospero test trace files are here: https://github.com/sstsimulator/sst-downloads/releases under "SST Test Support Files"

dmukherj09, 2023-07-17:
I was able to use those trace file in the prospero run test, I have one more  question -- can you explain me the format of the given trace file ?
example -- "1 W 140734192927288 8"

1 - ?
W - write
140734192927288 -?
8 - ?

dmukherj09, 2023-07-17:
Also I have another trace file which has the following format --
example -- "0 R 7d7134200 123"

where
0 - CPU number (there are 4)
R - read
7d7134200 - address
123 - data

I can convert my trace file to the required trace file using a python script after I get to know the format consumed by prospero but how to use the CPU number thing? is there a way to also give the CPU number as the insput to prospero?

gvoskuilen, 2023-07-20:
The format is "cycles type address length" where:
cycles: minimum cycle count at which the request can be issued
type: R (read) or W (write). Not case-sensitive.
address: memory address
length: number of bytes to read or write

Each core in prospero is its own component. So to do multiple cores you would instantiate multiple prospero CPUs, and give each one its own trace file.

dmukherj09, 2023-07-21:
I understand. Just one more follow-up questions, can you tell me, is there a way to get the average bandwidth for each cache and memory component. I couldn't find that in the dumped stats.csv. Also can you point me where i can get to know what each stats mean in the stats.csv, some of them I cannot understand what it means.

For example --

l1cache, GetXMiss_Arrival, , Accumulator, 244758996328, 0, 15192479, 15192479, 15192479, 1, 1
l1cache, GetSXMiss_Arrival, , Accumulator, 244758996328, 0, 0, 0, 0, 0, 0
l1cache, GetXMiss_Blocked, , Accumulator, 244758996328, 0, 2838, 2838, 2838, 1, 1
l1cache, GetSXMiss_Blocked, , Accumulator, 244758996328, 0, 0, 0, 0, 0, 0

gvoskuilen, 2023-08-11:
For the stats, use '$ sst-info memHierarchy' to locate the component/subcomponent and description of each stat. 'GetX' is an exclusive request for a block, 'GetS' is a non-exclusive request. 

For bandwidth, there's no direct stat that counts it. You'd need to add up the number of accesses and multiply by the size of the access to get the total volume. I'm going to mark this issue for a feature add since size of access isn't always a constant.

dmukherj09, 2023-08-14:
Yes adding the bandwidth metric would be great, thank you, for now I'm doing similar setup to what you mentioned to calculate the bandwidth.