sstsimulator/sst-elements #2175: Is it possible to directly connect CPU to rounter?

ideal8901, 2023-07-18:
Is it possible to directly connect CPU to rounter? 

When I do that I got error from merlin/interfaces/portControl.cc
"Error during PortControl protocol initialization.  The most likely cause of this is connecting an endpoint to a router port expecting to be connnected to another router."

Thank you!

gvoskuilen, 2023-07-20:
Is the goal to connect a CPU directly to the network rather than to say memory/cache first? If so, you can connect a CPU directly to a network by having it load a LinkControl (which implements SimpleNetwork) for the appropriate network model. For example, many of the included CPU libraries can load a series of subcomponents which implement interfaces that enable that connection. Specifically StdMem (memHierarchy.standardInterface) -> MemNIC (memHierarchy.MemNIC) -> LinkControl (merlin.linkcontrol).  Take a look at sst-elements/src/sst/elements/memHierarchy/tests/testIface-nic.py

ideal8901, 2023-08-01:
I see. 
I took a look at sst-elements/src/sst/elements/memHierarchy/tests/testIface-nic.py
But the problem unlike the config file, I am using mirandaCPU for these. Is it possible to use mirandaCPU instead of using memHierarchy.standardCPU?

Thank you!

ideal8901, 2023-08-02:
Thank you! I tried to connect my mirandaCPU with router based on the configuration file. 
However, when I tried to run code with my custom generator, it says it cannot find destination.
But the thing is the configuration runs with L1, L2 cache (and without StdMem,MemNiC) between mirandaCPU and router with the same configuration. 
Below is the error message. Do you have any idea for this problem?
------------------------------------

FATAL: mirandaCPU_0:memory:memlink (MemNICBase) cannot find a destination for address 2e0ec000
Known destinations:
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x409) [0x55b583d7ec49]
    1 : /home/jupark/sst/snoop_filter_branch/sst-home/sst-elements/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy10MemNICBase20getTargetDestinationB5cxx11Em+0x3ee) [0x7f264027c77e]
    2 : /home/jupark/sst/snoop_filter_branch/sst-home/sst-elements/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy17StandardInterface17MemEventConverter7convertEPNS_10Interfaces11StandardMem5WriteE+0x2a1) [0x7f26402d2241]
    3 : /home/jupark/sst/snoop_filter_branch/sst-home/sst-elements/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy17StandardInterface4sendEPNS_10Interfaces11StandardMem7RequestE+0x24) [0x7f26402d4704]
    4 : /home/jupark/sst/snoop_filter_branch/sst-home/sst-elements/lib/sst-elements-library/libmiranda.so(_ZN3SST7Miranda13RequestGenCPU12issueRequestEPNS0_15MemoryOpRequestE+0x1f9) [0x7f260023c3b9]
    5 : /home/jupark/sst/snoop_filter_branch/sst-home/sst-elements/lib/sst-elements-library/libmiranda.so(_ZN3SST7Miranda13RequestGenCPU9clockTickEm+0x489) [0x7f260023d819]
    6 : sst(_ZN3SST5Clock7executeEv+0x210) [0x55b583d2dc60]
    7 : sst(_ZN3SST15Simulation_impl3runEv+0x213) [0x55b583d9e933]
    8 : sst(+0xe97c0) [0x55b583d177c0]
    9 : sst(main+0x1497) [0x55b583cf9537]
   10 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f264e17fbf7]
   11 : sst(_start+0x2a) [0x55b583d1689a]

ideal8901, 2023-08-02:
I can observe the same problem in sst-elements/src/sst/elements/memHierarchy/tests/testIface-nic.py, 
when I removed l1/l1_nic in the configuration.

gvoskuilen, 2023-08-11:
The "group" parameters define which components on the network are sources/destinations of others. By default, "group 0" is a source for "group 1" which is a source for "group 2", etc. So try changing the mem_nic's "group" parameter to 1 so that it becomes the destination for cpu accesses.

closed 2023-08-15 by ideal8901