sstsimulator/sst-elements #2177: Explanation of the statout.csv metrics

dmukherj09, 2023-07-20:
Can someone help me in understanding or point me in the direction of the document where I can get the information on various metrics dumped by the SST?

ComponentName|StatisticName|StatisticSubId|StatisticType|SimTime|Rank|Sum.u64|SumSQ.u64|Count.u64|Min.u64|Max.u64

**cpu | read_reqs
cpu	| write_reqs
cpu	| custom_reqs
cpu	| split_read_reqs
cpu	| split_write_reqs
cpu	| split_custom_reqs
cpu	| cycles_with_issue
cpu	| cycles_no_issue
cpu	| total_bytes_read
cpu	| total_bytes_write
cpu	| total_bytes_custom
cpu	| req_latency
cpu	| time
cpu	| cycles_hit_fence
cpu	| cycles_max_issue
cpu	 | cycles_max_reorder
cpu	| cycles
l1cache	| prefetch_opportunities
l1cache	| prefetches_issued
l1cache	| prefetches_canceled_by_page_boundary
l1cache	| prefetches_canceled_by_history
l1cache	| Prefetch_requests
l1cache	| Prefetch_drops
.....**

dmukherj09, 2023-07-26:
Thank you, I'll check the links out

gvoskuilen, 2023-08-11:
Also, if you want to know more about a specific statistic dumped by a specific component (e.g., "cpu | cycles"), lookup that component type or library using the 'sst-info' utility. Such as `$ sst-info vanadis`. Components with subcomponents may also report statistics from their subcomponents so lookup any subcomponents as well if you can't find the statistic description with the main component.

dmukherj09, 2023-08-14:
That is a great suggestion, thank you!!