sstsimulator/sst-elements #2178: StdMem for atomic memory operations without cache

mrutt92, 2023-07-21:
Hi,

Is it possible to use the standardInterface in memHierarchy for atomic memory ops? I noticed the ReadLock and WriteUnlock commands in stdMem.h but they don't seem to guarantee exclusivity when I connected a standardInterface to a memoryController.

If I wanted to add this functionality without resorting to a cache, what would be the right way to do that?

mrutt92, 2023-07-25:
Any ideas on this?

gvoskuilen, 2023-08-11:
@mrutt92 memHierarchy implements the atomicity logic for ReadLock/WriteUnlock in the L1 caches only. If you wanted to add in that functionality, look at memHierarchy/memBackend/memBackendConvertor.cc::handleMemEvent() and the function it calls, memHierarchy/memBackend/membackendConvertor.h::setupMemReq(). This is where the logic for how and when the memory controller issues a request is implemented.