sstsimulator/sst-elements #2179: SST-Elements v13.0.0_Final Release version compilation problem

qizhi45, 2023-07-22:
New Issue for sst-elements
--------------------------
1 - Detailed description of problem or enhancement

SST-Elements fails to compile due to missing file extension.
Currently what I observe is:
Missing in:
1. sst/elements/memHierarchy/membackend/simpleMemScratchBackendConverto
2. sst/elements/mercury/operating_system/process/compute_scheduler_fwd.
3. sst/elements/mercury/operating_system/process/simple_compute_schedul

Maybe there are more files that are problematic.
These files are overwritten/renamed after unzip due to the missing suffix, which leads to compilation errors.

2 - Describe how to reproduce

Download SST Elements 13.0.0 from http://sst-simulator.org/SSTPages/SSTMainDownloads/
or https://github.com/sstsimulator/sst-elements/releases/tag/v13.0.0_Final
Released Date: 2023-APR-28
SHA1 Checksum: 1245fca5dbecd51a79e7ac330321324d86e4405e

3 - What Operating system(s) and versions 

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
SHA1 Checksum: 1245fca5dbecd51a79e7ac330321324d86e4405e

6 - Fill out Labels, Milestones, and Assignee fields as best possible

jpkenny, 2023-07-24:
Something on your system must be causing corruption during the extraction. The file with that checksum extracts fine on my local system and on all of our test platforms.

qizhi45, 2023-07-25:
oh, you are right, It's a bug with opening tar.gz files using some decompression Program.
Thanks for your response.

closed 2023-07-25 by qizhi45