sstsimulator/sst-elements #2181: Question on Tricount Motif

AKKamath, 2023-07-25:
I noticed that the documentation for Tricount mentioned that it was a replica of the SHAD/GMT implementation of Triangle Counting. 

From my understanding of the motif, it assumes that the input graph is undirected while the SHAD/GMT implementation is based on any type of graph, directed or undirected. 

I was wondering whether this difference was intentional, or if this is a bug.

jpkenny, 2023-09-01:
Are you referring to the appearance of
```
 if (src > dst) std::swap(src, dst);
```
in embertricount_setup?

If so, that is taken straight from the SHAD reference implementation.
As far as I know the motif replicates the SHAD code.
I agree the above swap would make for a strange directed graph.

AKKamath, 2023-09-05:
Yeah, that's what I was curious about. Thanks for the response!

closed 2023-09-05 by AKKamath