sstsimulator/sst-elements #2187: "UNABLE TO PROCESS LIBRARY = Merlin" , when implementing new topology

ziyuezzy, 2023-07-31:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

Dear developers of SST,

I am trying to implement new topologies in SST-Merlin. Three new files are added to the sst-element source file:
sst-elements-src/src/sst/elements/merlin/topology/new_topo.h
sst-elements-src/src/sst/elements/merlin/topology/new_topo.cc
sst-elements-src/src/sst/elements/merlin/topology/pymerlin-topo-new_topo.py

These files are added to 'libmerlin_la_SOURCES' in file "Makefile.am" and "Makefile.in" , so as for generating '.lo' and '.plo' files.

The following code is also added to "merlin.cc":
char pymerlin_topo_new_topo[] = {
#include "topology/pymerlin-topo-new_topo.inc"
    0x00};

'make' Compiling went well, no error occurred. 'make install' installation also did not show error.
However after the compilation and installation, 'Merlin' is not recognized by SST anymore, even though file "sst-elements/sst-elements-install/lib/sst-elements-library/libmerlin.so" exists. (screenshot below)
[image]

Here is the error:
**** WARNING - UNABLE TO PROCESS LIBRARY = Merlin
PROCESSED 0 .so (SST ELEMENT) FILES FOUND IN DIRECTORY(s) /users/ziyzhang/SST/sst-core/build/../sst-core-install/lib/sstcore:/users/ziyzhang/SST/sst-elements/build/../sst-elements-install/lib/sst-elements-library
Filtering output on Element = "Merlin"

I have tried to compile and run the original version of Merlin (thus without my modifications of the new topologies), the error disappears.

Do you have any idea what is the problem here?

Best regards,
Ziyue.

2 - Describe how to reproduce
The modified version of sst-element is here: https://github.com/ziyuezzy/sst-elements/tree/topo_from_graph

3 - What Operating system(s) and versions 

ubuntu 20

4 - What version of external libraries (Boost, MPI)

Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.1' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-Av3uEd/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

original sst-core, 
Modified sst-element (only changed Merlin): https://github.com/ziyuezzy/sst-elements/tree/topo_from_graph

6 - Fill out Labels, Milestones, and Assignee fields as best possible

Lables: Merlin, Help Wanted, Compilation

feldergast, 2023-07-31:
When the library isn't recognized, it usually means there's a missing symbol in the library.  Nine times out of ten this ends up being a destructor that's missing.  Sometimes you'll get a message about the missing symbol, but it depends on the OS and sometimes the version of Python you compiled against.  I'll take a quick look at the code and see if something sticks out.

feldergast, 2023-07-31:
I think the likely issue is MAX_PATH_LENGTH.  It's declared as static in the header file, but there is no instantiation of the variable in the .cc file, so the symbol is never actually allocated.  This would likely cause the behavior you're seeing as the library would have a missing symbol and would fail to load.

ziyuezzy, 2023-07-31:
Hi,
Thank you so much to look into this! 
Indeed by properly instantiating the static variable, the library got recognized.

closed 2023-07-31 by ziyuezzy