sstsimulator/sst-elements #2188: Explanation of logical groups in Merlin?

jakess22, 2023-07-31:
New Documentation question for sst-elements/Merlin
--------------------------

1 - Detailed description of problem or enhancement
Does anyone know how logical groups can be used with Merlin topologies and/or Merlin's hr_router? I am looking into how to change bandwidth for hr_router during simulation, and the code block below seems to indicate this is possible with logical groups, but I have been unable to find (1) how to implement them and (2) how they affect hr_router state?

2 - from hr_router.cc

// Link BW default.  Can be overwritten using logical groups
    std::string link_bw_s = params.find<std::string>("link_bw");
    UnitAlgebra link_bw(link_bw_s);

3 - What Operating system(s) and versions 
NA

4 - What version of external libraries (Boost, MPI)
NA

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
NA

6 - Fill out Labels, Milestones, and Assignee fields as best possible

feldergast, 2023-08-03:
Can you give more detail on what you're trying to accomplish?  If you just want to set the bandwidths for all ports to the same value, you would set the link_bw parameter on the router.  Or it sounds like maybe you want to change the bandwidth during simulation?  Doing that would require a change to the router models to enable, since that is done after configuration time.  I believe that has been done before, but I'm not sure if the support is still there.  It was for a one-off paper a student was doing.

As far as logical ports: logical ports was a feature designed to allow you to statically set different types of ports to different values.  Only the dragonfly topology uses the feature and it has host, group and global logical port designations.  Unfortunately, while the underlying simulation models support this, the new merlin python module does not yet support this feature, making it difficult to configure.  If you are interested in setting different bandwidths on the different types of ports, I will need to augment the python model to support this.  This is not a feature anyone has wanted to use since we updated the merlin python module and there are some challenges for getting it working, hence it was never implemented.

jakess22, 2023-08-08:
Hi! Sorry for the delay in response. Changing bandwidth during simulation is the feature I am looking into. What would be needed to update the router models?

jakess22, 2023-08-08:
We are looking to model new optical switches, which support configurable bandwidth at all ports. Does this require further updates to support routers with ports of different bandwidths concurrently?

feldergast, 2023-08-10:
Actually changing the bandwidth on the link is pretty simple.  The bandwidth is controlled using the output_timing link in PortControl and LinkControl.  The time base for the link is set to be the serialization delay of a single flit, and each time a packet is sent, a "wake-up" event is sent on the output_timing link with an added latency equal to the number of flits (which gets multiplied by the time base), which means the wake-up event will be delivered at the time corresponding to when the current packet finishes sending.  If you change the timebase, you will effectively change the bandwidth on the link.

The harder part is deciding how you want to control when the changes happen and plumbing in all the changes so that the component knows when to switch bandwidths.  If you can tell us a little more about how you want to drive the changes, we might be able to offer some more insight into how to go about it.  If you want to connect offline, you can email wg-sst@sandia.gov and we'll connect you with the right people.

closed 2023-10-11 by hughes-c