sstsimulator/sst-elements #2189: Start stat collection at specified simulation time

plafratt, 2023-07-31:
Is there a way to start stat collection at a specific simulation time?

I have a simulation that has about 500 us of initialization phase during which the stats are not of interest to me. I wanted to ask if there is a way to tell SST to exclude this period of the simulation from stats collection.

Any help is appreciated.

feldergast, 2023-08-01:
It is possible to start and stop stats at specific times.  This is controlled by parameters to the enableStatistics calls.

More information on statistics can be found here:

http://sst-simulator.org/sst-docs/docs/config/stats/overview

Look for startat and stopat on the overview page, and examples can be found in the specific enableStatistics calls.

plafratt, 2023-08-01:
Thank you! I was able to find the relevant info.

plafratt, 2024-02-29:
I had a follow up question to this - is it possible to start and stop stat collection for a specific region of code?

I think one way of doing it is to run the simulation once to determine the times at which the region starts and ends, then set the `start_at` and `stop_at` parameters to correspond to those times. But, there are some cons to that approach (e.g. it involves running the simulation twice).

Is there possibly a better way?

gvoskuilen, 2024-02-29:
Vanadis does not have this capability, but SST Core supports it (with some limitations) and Ariel does this today. When Ariel detects a particular dummy function call in the simulated application, it calls this component API function: http://sst-simulator.org/sst-docs/docs/core/component/stat/performGlobalStatisticOutput. 

Caveat: the output is limited to statistics belonging to components on the same rank/thread as the component that triggered the output. We have some thoughts on how to make it truly global which involve deferring the dump until the simulation's next sync point.

Could you create an enhancement issue for vanadis to track this feature? If you need it sooner and end up implementing it, we can merge the capability in. Otherwise, we'll add it to our backlog for features.

plafratt, 2024-02-29:
Thank you for the helpful information. I will create the enhancement issue as requested and post un update there if we end up implementing it.

closed 2024-02-29 by plafratt

plafratt, 2024-03-01:
As requested, I created an enhancement request #2325. If we are able to implement this feature internally, I will post an update there. Thank you again.