sstsimulator/sst-elements #2194: How to set debug level of StandardInterface?

plafratt, 2023-08-09:
For a small number of requests in my simulation, I am seeing a huge latency (>1.25 us) between the VanadisBasicLoadStoreQueue (at the `memInterface->send()` statement [here](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/vanadis/lsq/vbasiclsq.h#L867)) and the L1 cache (when it receives the request in `clockTick()` [here](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/cacheController.cc#L155)). I am trying to figure out what is causing that latency.

I am trying to debug it, and I'd like to see this debug statement: [link](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/standardInterface.cc#L190), but I can't figure out how to set the debug level for StandardInterface.

I tried doing this, but it didn't work.

```
l1dcache_2_cpu     = cpu_l1dcache.setSubComponent("cpulink", "memHierarchy.MemLink")
l1dcache_2_cpu.addParams( { "debug" : 1, "debug_level" : 10 } )
```

I did compile with debug symbols (I am able to see the debug output from components).

Any help is appreciated.

hughes-c, 2023-08-09:
You need to turn on debugging in vanadis and the L1. Debugging in vanadis and the lsq subcomponent are controlled by the _verbose_ flag. However, send() should hit the L1 immediately.

hughes-c, 2023-08-09:
`E: 1001                 1                    df_l1                Event:Recv    (<10,0> GetS Src: df_0:iface Dst: df_l1 Rq: df_0:iface Tid: 0 Flags: [] Addr: 0x0 (G) Data: F VA: 0x0 IP: 0x0 Size: 8 Prf: F)`

Then L1 _access_latency_cycles_ later:
E: 2000                 2                    df_l1                Event:New     (<10,0> GetS Src: df_0:iface Dst: df_l1 Rq: df_0:iface Tid: 0 Flags: [] Addr: 0x0 (G) Data: F VA: 0x0 IP: 0x0 Size: 8 Prf: F)
M: 2000                                      df_l1                     MSHR:InsEv    0x0                Sz: 1      (<10,0>, pos=0)
M: 2000                                      df_l1                     MSHR:Profile  0x0                Sz: 1     
M: 2000                                      df_l1                     MSHR:InProg   0x0                Sz: 1     
C: 2000                 2                    df_l1                0    GetS          0x0                <10,0>          I      IS     Forward    (miss)          LLSC: N LLSCTime: 0 LLSCTid: [] Lock: (0,0)

gvoskuilen, 2023-08-11:
To enable debug on the StandardInterface, you need to add the debug parameters to the StandardInterface which means if you are not already, you must also explicitly instantiate the interface in your input file. Something like this (I borrowed from memHierarchy/tests/sdl9-1.py):

```
cpu = sst.Component("core", ...)

iface = cpu.setSubComponent("memory", "memHierarchy.standardInterface")
iface.addParams({ "debug" : 1, "debug_level" : 5 })
```

plafratt, 2023-08-11:
Thanks! I will give that a try.

closed 2023-08-12 by plafratt