sstsimulator/sst-elements #2224: Pipeline stages sequence in Vanadis

Connie120, 2023-09-25:
Hello,

I'm wondering why the simulation sequence for the pipeline stages in Vanadis is Fetch->Decode->Issue->Execute->Retire, rather than Retire->Execute->Issue->Decode->Fetch, which is the other way around. 

I tried to run a simple program using the latest Vanadis in the devel branch and printed out the ROB states:

[image]

This is the first cycle (cycle 13590) that Vanadis starts to simulate the program. As from the screenshot, the instruction PCADDI64 (Addr 1013e) is decoded and issued in the same cycle, and it's sent to the execution unit. The reason why it's not retired is that it hasn't finished execution. 

So I tried to invert the sequence of the stages and printed out the the ROB states again, also cycle 13590:

[image]

Here the instruction is only decoded but not issued, and it's issued in the next cycle (cycle 13591):

[image]

Please let me know if I understand the code incorrectly and thanks for the help!

hughes-c, 2023-10-11:
@Connie120 After internal discussions, we think that swapping the pipeline order is a viable solution. You're welcome to submit a PR to fix this or we can work on it.

Connie120, 2023-10-11:
Created a PR in #2240, please check. Thanks!