sstsimulator/sst-elements #2256: Want to change MemLink to MemNIC

JoongunPark, 2023-10-23:
Hi, I am currently trying to make new component between router and memory controller. 

MC - MemLink - (My component) - MemNIC - Router. 

It works as it is but I want to change MemLink to MemNIC. 
Is there any specific sub component or method to do that? 

Other wise, how can I convert MemLink to MemNIC? 

Thank you!

gvoskuilen, 2023-10-23:
Is your goal to have
MC - MemNIC - Router
(your component) - MemNIC - Router

or something else?

Generally, to switch a MemLink for a MemNIC, in the python, you'd use this:
```
memnic = mem.setSubComponent("cpulink", "memHierarchy.MemNIC")
link.connect( (memnic, "port", latency), ( ... ) ) # Instead of linking mem
```

But depending on which components talk to others on the network, you will also need to set the source and/or destination parameters on the MemNICs so that they correctly forward events.

JoongunPark, 2023-10-23:
Oh my goal is to have like below.  
MC - **MemNIC** - (My component) - MemNIC - Router 

Currently it is working like below. 
MC - **MemLink** - (My component) - MemNIC - Router 

Thank you!

gvoskuilen, 2023-10-23:
The MemNIC has two tasks (1) look up the network address for the destination of a memory request/response according the event's routing address and (2) wrap the memory event in a network event with that destination configured and send it to the MemNIC's LinkControl subcomponent (e.g., merlin.linkcontrol or kingsley.linkcontrol). Linkcontrols expect to be connected to a network. So, I think your configuration as written will not work. You cannot use a network link (memNIC) on a non-network connection.

JoongunPark, 2023-10-23:
I see... Thank you so much for the comments. Those really help me!

JoongunPark, 2023-10-23:
Then is there a way to convert MemNIC to MemLink? 
For example, I want to make directory controller only have Memlinks. 

Memory Controller - Memlink - Directory Controller - **Memlink** - Component - MemNIC - Router

gvoskuilen, 2023-10-23:
Yes. In the python, set the directory controller's cpulink to a MemLink. 
```python
dc = sst.Component("directory", "memHierarchy.DirectoryController")
dc_cpulink = dc.setSubComponent("cpulink", "memHierarchy.MemLink") # Connect this to your component
dc_memlink = dc.setSubComponent("memlink", "memHierarchy.MemLink") # Connect this to the memory controller

link_mem_dc.connect( (dc_memlink, "port", latency), (<mem controller port>) )
link_dc_comp.connect( (dc_cpulink, "port", latency), (<your component port>) )
```

JoongunPark, 2023-10-23:
Thank you so much! 
I will try that.

closed 2023-10-23 by JoongunPark