sstsimulator/sst-elements #2258: Error: Found destinations on the network with overlapping address regions. Cannot generate routing table.

JoongunPark, 2023-10-23:
Hello, though I don't register region for L2 cache, I met the below error.

CPU          MC (Memory Controller)             MC
  l                  l                                                  l
L1/L2        DC (Directory Controller)             DC
   l                 l                                                  l
Router  -  Router    ------------------------- Router    ....
 
Could you please let me know how to handle this error? 
Thank you!

```
FATAL: custom_component, Error: Found destinations on the network with overlapping address regions. Cannot generate routing table.
  Destination 1: Name: l2_cache_0 Addr: 3 ID: 2 Region: Start: 0 End: 0xffffffffffffffff InterleaveSize: 0 InterleaveStep: 0
  Destination 2: Name: directory_ctrl_3 Addr: 1 ID: 3 Region: Start: 0 End: 0x63fffff InterleaveSize: 0 InterleaveStep: 0
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x409) [0x5632a7699c49]
    1 : /home/jupark/sst/snoop_filter_branch/sst-home/sst-elements/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy10MemNICBase5setupEv+0x6ba) [0x7f53fcfe7c8a]
    2 : sst(_ZN3SST15Simulation_impl5setupEv+0x7e) [0x5632a76b837e]
    3 : sst(+0xe979d) [0x5632a763279d]
    4 : sst(main+0x1497) [0x5632a7614537]
    5 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f540ef27c87]
    6 : sst(_start+0x2a) [0x5632a763189a]
```

```
coherence_protocol = "MESI"

l1_cache_params = {
    "debug": 0,
    "cache_frequency": core_clock,
    "cache_size": "64KiB",
    "associativity": 8,
    "access_latency_cycles": 20,
    "L1": 1,
    "cache_line_size": cache_line_size,
    "coherence_protocol": coherence_protocol,
    "cache_type": "inclusive",
    "mshr_num_entries": 48,
    "mshr_latency_cycles": 4,
    "memNIC.network_bw": ring_bandwidth,
}

l2_cache_params = {
    "verbose": 0,
    "cache_frequency": core_clock,
    "cache_size": "256KiB",
    "associativity": 16,
    "access_latency_cycles": 100,
    "L1": 0,
    "cache_line_size": cache_line_size,
    "coherence_protocol": coherence_protocol,
    "cache_type": "inclusive",
    "mshr_num_entries": 256,
    "mshr_latency_cycles": 8,
    "memNIC.network_bw": ring_bandwidth,
    "num_cache_slices": num_pes,
    "slice_allocation_policy": "rr",
}

# PE ------------------------------------------------------------------
print("\tBuilding mirandaCPU_%d" % (pe_id))
pe = sst.Component("mirandaCPU_%d" % (pe_id), "miranda.BaseCPU")
pe.addParams(pe_params)
#----------------------------------------------------------------------

# L1 ------------------------------------------------------------------
print("\tBuilding l1_cache_%d" % (pe_id))
l1_cache = sst.Component("l1_cache_%d" % (pe_id), "memHierarchy.Cache")
l1_cache.addParams(l1_cache_params)
#----------------------------------------------------------------------

# L2 ------------------------------------------------------------------
print("\tBuilding l2_cache_%d" % (pe_id))
l2_cache = sst.Component("l2_cache_%d" % (pe_id), "memHierarchy.Cache")
l2_cache.addParams(l2_cache_params)
l2_cache.addParams({"slice_id": pe_id})
#----------------------------------------------------------------------

# Custom Random Generator ------------------------------------------------------
gen = pe.setSubComponent("generator", "miranda.CustomRandomGenerator")
gen.addParams({
    "avoid_addr_range_min_value" : args.node4minAddr,
    "avoid_addr_range_max_value" : args.node4maxAddr,
    "max_address" : total_memory_size_in_MB,
    "verbose" : 0,
    "count" : args.packet_count,
})
#----------------------------------------------------------------------

print("\tCreating mirandaCPU_%d_l1_cache_%d_link"%(pe_id, pe_id))
connect("mirandaCPU_%d_l1_cache_%d_link"%(pe_id, pe_id),
        pe, "cache_link",
        l1_cache, "high_network_0",
        ring_latency).setNoCut()

print("\tCreating l1_cache_%d_l2_cache_%d_link"%(pe_id, pe_id))
connect("l1_cache_%d_l2_cache_%d_link"%(pe_id, pe_id),
        l1_cache, "low_network_0",
        l2_cache, "high_network_0",
        ring_latency).setNoCut()

print("\tCreating l2_cache_%d_rtr_%d_link"%(pe_id, rtr_id))
connect("l2_cache_%d_rtr_%d_link"%(pe_id, rtr_id),
        l2_cache, "directory",
        router_map[rtr_name], "port2",
        ring_latency)
```

gvoskuilen, 2023-10-23:
This means the customcomponent MemNIC has two destinations registered for the same address so it won't know where to send a request for that address. The fix is to set the group/source/destination IDs on the MemNICs so that they can figure out the logical routing path through your architecture.

Assuming the L2 directs requests to the dirs and your custom component, and you have all the group/src/dest parameters are their default values, try this:
```python
customComponent_memnic.addParam("group", 3);
```

JoongunPark, 2023-10-23:
It worked! Thank you so much!

closed 2023-10-23 by JoongunPark