sstsimulator/sst-elements #2263: How to check the packet's source location?

JoongunPark, 2023-10-25:
Hello,

I have set up the following topology and it's functioning correctly:

```
Copy code
Router - L2/L1 - CPU1
|
Router - L2/L1 - CPU2
|
Router - Directory Controller - Memory Controller1
|
Router - **(custom_component)** - Directory Controller - Memory Controller2
```

I'd like to determine if the (custom_component) can identify the source CPU by inspecting the packets to Memory Controller2.
Thank you!

gvoskuilen, 2023-11-14:
Yes, MemEvent events have a field `rqstr_` which contains the name of the memHierarchy.standardInterface that caused the event to be generated. It will look something like `CPU1:memory` or `CPU2:memory` (format: cpu_name:subcomponent_slot_name).

JoongunPark, 2023-11-15:
Thank you! I will check that out.

closed 2023-11-15 by JoongunPark