sstsimulator/sst-elements #2264: How to sst runstream.py?

gihyeonjeon, 2023-10-26:
Hi, this is my first time for sst and github.. and my english skill would be not good.. 

I want to study about scratchpad memory, so i need some examples about scratchpad or ariel...
So i found runstream.py from ariel/frontend/simple/examples/stream and i have entered "sst runstream.py" in terminal.
But there is an error message, "Error: Loading fronted subcomponent failed. If Ariel was not build with Pin, user must supply a custom fronted in the input file.".
I can't deal with it.
Can you give me advice plz??..

hughes-c, 2023-10-26:
Hi, you likely need to install Pin and build SST with Pin. See the instructions here:
https://sst-simulator.org/SSTPages/SSTBuildAndInstall_13dot0dot0_SeriesAdditionalExternalComponents/#intel-pin-tool-326-98690

gihyeonjeon, 2023-10-27:
Really thank you!
But i have another problem...

I cannot install Pin.
I installed Intel PIN ver. 3.26 and set the home directory environment successfully.
And i changed directory to ~/scratch/src/sst-elements-library-13.0.0!
But when i configured SST-Elements such as "./configure --with-pin=$PIN_HOME" and "./configure --prefix=$PIN_HOME --with-pin=$PIN_HOME", i got error message, "configure: error: PIN was requested but not found"

I'm sorry but it is hard for me...

gihyeonjeon, 2023-10-27:
I have solved this problem by changing "--with-pin=$HOME/scratch/src/pin-3.26~". I don't know that this is right way but i found a directory of pin-3.26 under the path $HOME/local/packages/. 

And i entered a command "sst runstream.py" but i got a new meassage "FATA: Pin3Fronted[frontend/pin3/pin3fronted.cc:373:forkPINChild] Launching trace child failed! Child Exited with status 2".

My installing way is not good or what?..

closed 2023-11-05 by gihyeonjeon