sstsimulator/sst-elements #2266: Is my understanding about Scratchpad.cc correct?..

gihyeonjeon, 2023-10-31:
Hi!
In the code provided, if the responsesPerCycle value is set to 0, the responseThisCycle is initialized to 1. With the subsequent increment of responseThisCycle, its value becomes 2. This would mean that no matter how many times it iterates, responseThisCycle will never equal responsesPerCycle, resulting in an infinite loop. Is my understanding of this correct?

    uint32_t responseThisCycle = (responsesPerCycle_ == 0) ? 1 : 0;
    while (!procMsgQueue_.empty() && procMsgQueue_.begin()->first < timestamp_) {
        MemEventBase * sendEv = procMsgQueue_.begin()->second;

        if (is_debug_event(sendEv)) {
            debug = true;
            dbg.debug(_L4_, "E: %-20" PRIu64 " %-20" PRIu64 " %-20s Event:Send    (%s)\n",
                    getCurrentSimCycle(), timestamp_, getName().c_str(), sendEv->getBriefString().c_str());
        }

        linkUp_->send(sendEv);
        procMsgQueue_.erase(procMsgQueue_.begin());
        responseThisCycle++;
        if (responseThisCycle == responsesPerCycle_) break;
       }

gvoskuilen, 2023-11-01:
Setting responsesPerCycle_ to 0 indicates that every ready event in the procMsgQueue_ should be drained every cycle. So the break statement will not be taken but the loop is not infinite because procMsgQueue_ is being drained and eventually the while condition will not be met.

gihyeonjeon, 2023-11-05:
Really thank you!!

closed 2023-11-05 by gihyeonjeon