sstsimulator/sst-elements #2276: Ember Tricount - Segmentation fault

AKKamath, 2023-11-29:
With certain configurations, I'm getting a segmentation fault in the Tricount motif.
I've isolated the error to this line:
https://github.com/sstsimulator/sst-elements/blob/d723ad035dd83c7a1957c9a9b90b7fc73733c7d8/src/sst/elements/ember/mpi/motifs/embertricount.cc#L352C9-L352C9

Right now the code is:
`if (i > taskStarts_.size() || j > taskStarts_.size() ) skip = true;`

I believe this should instead be:
`if (i >= taskStarts_.size() - 1 || j >= taskStarts_.size() - 1 ) skip = true;`

This is because the [line 361](https://github.com/sstsimulator/sst-elements/blob/d723ad035dd83c7a1957c9a9b90b7fc73733c7d8/src/sst/elements/ember/mpi/motifs/embertricount.cc#L361C8-L362C73) has the following accesses:
```
first_i = taskStarts_[i];      // first vertex of partition i
last_i  = taskStarts_[i + 1];  // first vertex of partition i + 1
```
If i is equal to taskStarts_.size() - 1, then taskStarts_[i + 1] is an out-of-bounds access.

jpkenny, 2023-11-29:
You're right. I believe I was counting on line 353 to catch index equal to taskStarts_.size() - 1, but upon reflection it won't do that for all cases. I can clean this up a bit and put in a patch.

jpkenny, 2023-11-29:
Addressed by PR #2277
@AKKamath try the latest on devel

hughes-c, 2023-12-01:
@AKKamath Can we close this?

AKKamath, 2023-12-01:
Yup, seems fine now. Thanks!