sstsimulator/sst-elements #2287: Firefly: Module load failed - pointers for debugging

saichennaintel, 2023-12-12:
I'm currently trying to add nonblocking collectives into firefly. I have made some code modifications. While rebuilding firefly (using make all && make install), the build does not throw any errors however if I check the installation by running sst-info firefly. I get the following output:

[image]

As a result, while running a test configuration, it throws "No module named sst.firefly"
[image]
 
Any pointers on how to debug the issue? As it is not throwing any error during the build time, it makes it harder to find the root cause.

feldergast, 2023-12-13:
The most likely cause of this is a missing symbol in the libfirefly library (for some reason, it's most often a destructor).  Since you are building a dynamically loaded library, the linker doesn't error on missing symbols, but the loading code in the core will fail if there is a symbol referenced that it can't resolve.  Unfortunately, as hard as we've tried, we haven't been able to get more specific error messages for this case.

feldergast, 2024-03-27:
@saichennaintel Were you able to find the issue with loading?  If so, is it okay if we close this issue?

saichennaintel, 2024-04-02:
You can close the ticket. I have put this work on hold currently. I'll follow your pointers on probable suspects while debugging on I resume the work.

closed 2024-04-02 by saichennaintel