sstsimulator/sst-elements #2313: The control mechanism for the number of prefetch events in memHierarchy (variable outstandingPrefetches_)

zuiaikugua, 2024-02-02:
Why there is no decrement operation on the variable `outstandingPrefetches_` in memHierachy?

Recently I have been studying the perfetch mechanism in Cache. It is found that the variable **outstandingPrefetches_** is used in memHierarchy to control the number of perfetch events that can be issued. 
However, in memHierarchy, there is only increment operation for **outstandingPrefetches_**, and there is no decrement operation. This results in that even if the prefetch data has been returned to the cache, the new prefetch event cannot be delivered. This reduced perfecter capabilities.

gvoskuilen, 2024-02-02:
This is a bug, thank you for reporting it. We will fix it.

zuiaikugua, 2024-02-07:
Thanks for your quick bug fixing. I have read the commit code. But I still have a question: Why you don't add the prefetch control code in the `memHierarchy/coherencemgr/MESI_Private_Noninclusive.cc` ?

gvoskuilen, 2024-02-07:
The noninclusive private cache can't have its own prefetcher because it cannot tell if a locally generated prefetch is present in an upper level cache (e.g., a noninclusive/private L2 doesn't know what is in L1). Since the prefetch limiting only limits locally-generated prefetches and not prefetches initiated by other caches, that protocol doesn't need it.

zuiaikugua, 2024-02-08:
> The noninclusive private cache can't have its own prefetcher because it cannot tell if a locally generated prefetch is present in an upper level cache (e.g., a noninclusive/private L2 doesn't know [...]

OK, I see. Thanks for your reply.

PR #2315 (2024-02-06): memH: Fix bug in prefetch logic
Fixes #2313
closed 2024-02-08 by zuiaikugua