sstsimulator/sst-elements #2323: Errors when running ember motifs

ziyuezzy, 2024-02-22:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

**_Dear,_**
 
I am trying to run the classical ember+firefly+merlin stack. I could run some of the ember motifs such as Allreduce, Alltoall, Halo2D, etc. However, some ember motifs lead to errors that I don't find a direction to resolve, for example with ember/tests/dragon_128_allreduce.py as the sdl file:

Adding ep.addMotif("Halo3D") : 
the program stalls while the cpu occupancy of sst is still high. I suspect there is a deadlock?

Adding ep.addMotif("FFT3D") : 
Error message: `emberfft3d.cc:45: SST::Ember::EmberFFT3DGenerator::EmberFFT3DGenerator(SST::ComponentId_t, SST::Params&): Assertion 0 < m_data.nprow failed.)` Is this a bug in the FFT3D motif, or is it there some additional configurations needed?

Adding ep.addMotif("shmem*") :
Error message: `EmberEngine:getLib:99: Error: could not find shmem`
I wonder that do we need to install the real openSHMEM for running shmem-related motifs? or is there some library bugs issues here?

Overall, is there a documentation about all these motifs?

**_Thanks!
Best regards,
Z._**

2 - Describe how to reproduce

compile the following sst elements:
SST-ember
SST-firefly
SST-merlin
SST-Hermes

then run "sst .../sst-elements-src/src/sst/elements/ember/tests/dragon_128_allreduce.py", replace the ep.addMotif("Allreduce") with other correspodnmotifs.

3 - What Operating system(s) and versions 
ubuntu20.04LTS

4 - What version of external libraries (Boost, MPI)
mpirun (Open MPI) 4.0.3

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
original sst core and elemets
https://github.com/sstsimulator/sst-core
https://github.com/sstsimulator/sst-elements

6 - Fill out Labels, Milestones, and Assignee fields as best possible
Labels: SST-ember; help wanted

feldergast, 2024-02-22:
I'm not sure about the shmem motifs, but for halo3d and fft3d there are arguments that need to be supplied.  For halo3d, you are probably getting the default parameters, which are likely setting up a large problem so is taking a long time to run.  For fft3d, it must error if you don't specify a required parameter.

To get the parameters available, run sst-info ember.Halo3DMotif and look for the parameter listing and find the parameters that start "arg."  The actual argument passed to the motif does not include the "arg.".

For example, here is the listing of some of the parameters for halo3d:

         arg.nx: Sets the problem size in X-dimension  [100]
         arg.ny: Sets the problem size in Y-dimension  [100]
         arg.nz: Sets the problem size in Z-dimension  [100]
         arg.pex: Sets the processors in X-dimension (0=auto)  [0]
         arg.pey: Sets the processors in Y-dimension (0=auto)  [0]
         arg.pez: Sets the processors in Z-dimension (0=auto)  [0]

To use them in the motif call, you would use:

addMotif("Halo3D nx=16 ny=16 nz=16 pex=4 pey=4 pez=4")

I believe the nx, ny and nz parameters set the per PE problem size and the pex, pey and pez parameters set the decomposition and must multiply to be the number of nodes in the run.

You can use sst-info in a similar manner to get the information for other motifs.  The naming convention for the SubComponent name is ember.<MotifName>Motif.

feldergast, 2024-02-24:
Yes 18.4 mega seconds is the limit of the 64-bit number.  However, if you are hitting that for a motif, then there is either an incompatible set of inputs or there is a bug somewhere (or both). I've never looked at the code for FFT3D, so I'm not sure which of those it may be.

closed 2024-02-26 by ziyuezzy