sstsimulator/sst-elements #2342: PRI_ADDR is not available on all compilers

ukasz, 2024-03-28:
Usage of PRI_ADDR leads to compilation errors on "some" compilers.
**This macro should be set up manually if it is not defined by the compiler**, otherwise it is not defined and compiler complains about missing type after % sign:
```
./os/vappruntimememory.h:447:110: error: expected ')' before 'PRI_ADDR'
  447 |             CALL_INFO, 16, VANADIS_OS_DBG_APP_INIT, "-> Pushing %" PRIu64 " bytes to the start of stack (0x%" PRI_ADDR ") via memory init event..\n",
      |                        
```
Memhierarchy defines it manually, but other elements are not so smart.
Now there are so many usages of it that fixing it manually becomes annoying.

My OS: Ubuntu 22.04
GCC: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)

feldergast, 2024-04-02:
PRI_ADDR is defined in sst/core/interfaces/stdMem.h, which is included in the file giving you an error.  This was a relatively recent addition to stdMem.h; is it possible you're using a slightly older version of SST core? However, I'm not sure why this would affect only some compilers so it may be something else.

ukasz, 2024-04-02:
You are right it is a matter of using old rev core version, and switching to a newer one fixes the problem. However I still think that it would be better to define it by the elements if they want to use it. It breaks compatibility for no real reason as it is now.

feldergast, 2024-04-02:
The #define for PRI_ADDR is associated with "typedef uint64_t Addr" in stdMem.h (they are defined on consecutive lines in the header file).  If you are using the Addr typedef, then you also get the PRI code for it.  PRI_ADDR shouldn't be used unless you are specifying an Addr as the type.

For the line you are seeing an error on, the type of start_stack_address should maybe be Addr in the function definition.  So the issue is not necessarily in the use of PRI_ADDR in the print statement, but rather using Addr and PRI_ADDR consistently for the correct variables.  This consistency issue should be addressed.

jwilso, 2024-09-20:
@feldergast Does a new issue need to be created for StandardMem::Addr?