sstsimulator/sst-elements #2344: FATAL: (MemLink) cannot find a destination for address 0

Adrian-2105, 2024-04-02:
Hello, I am trying to update an external Component called `sstStonne` from SST 8.0.0 to SST 13.1.0. It's a simple Component which only connects to memory, thus I am updating from the `SimpleMem` interface to the `StandardMem` interface. However I am facing some problems and cannot figure out where the error is.

`sstStonne` requires a SubComponent, named `memory`, which implements the `SST::Interfaces::StandardMem` interface. The code compiles properly and is added to SST correctly. At execution time, it successfully loads the `memory` SubComponent and even gets to send a memory initialization request through `StandardMem::sendUntimedData` at the `init()` phase without getting any error. However, I am getting the next fatal error when trying to call `StandardMem::send()` in the 1st cycle of simulation:

```
FATAL: stonne:memory (MemLink) cannot find a destination for address 0
```

I guess I am missing some extra configuration, or maybe I am not using `StandardMem` properly, but I didn't find any further information about this in the documentation and the examples are working for me.

I attach further information next, hope it can be helpful. Thanks in advance!

-----

`sstStonne.h` (main Component)

```c++
class sstStonne : public SST::Component {
 public:
  {...}
  SST_ELI_DOCUMENT_PORTS()

  SST_ELI_DOCUMENT_SUBCOMPONENT_SLOTS({"memory", "The memory interface to use (e.g., interface to caches)", "SST::Interfaces::StandardMem"})
}
```

`SstMem.h` (ComponentExtension from `sstStonne`)

```c++
template <typename T>
class SstMem : public Memory<T>, private ComponentExtension {
 public:
  SstMem(SST::ComponentId_t id, SST::Output* output, SST::TimeConverter* timeConverter)
      : ComponentExtension(id),
        m_output(output),
        m_requestHandler(output, *this),
        m_memInterface(loadUserSubComponent<SST::Interfaces::StandardMem>(
          "memory", SST::ComponentInfo::SHARE_NONE, timeConverter, new SST::Interfaces::StandardMem::Handler<SstMem<T>>(this, &SstMem<T>::handleEvent))) {
    if (!m_memInterface) {
      m_output->fatal(CALL_INFO, -1, "Unable to load memory interface. Make sure you attach a SubComponent to the STONNE Component called \"memory\"\n");
    }
  }

  void init(uint32_t phase, std::uint64_t startAddr, const std::vector<std::uint8_t>& data) noexcept {
    if (phase != 0) return;

    m_memInterface->init(phase);
    auto initMemory = new SST::Interfaces::StandardMem::Write(startAddr, data.size(), data);
    m_memInterface->sendUntimedData(initMemory);
  }

  void load(uint64_t addr, DataPackage* pck) noexcept override {
    auto request = new SST::Interfaces::StandardMem::Read(addr, sizeof(T));
    m_memInterface->send(request);
  }

{...}
 private:
  // SST Memory Interface
  SST::Interfaces::StandardMem* m_memInterface;

  // SST Output
  SST::Output* m_output;
}
```

`sstStonne_test.py` (SST API, based on [`testStdMem.py` example](https://github.com/sstsimulator/sst-elements/blob/v13.1.0_Final/src/sst/elements/memHierarchy/tests/testStdMem.py) ; all connections seem to be configured properly)

```python
import sst

# Define SST core options
sst.setProgramOption("timebase", "1 ps")

statLevel = 16
max_addr_gb = 16
tile_clk_mhz = 1

# Define the simulation components
stonne = sst.Component("stonne", "sstStonne.MAERI")
stonne.addParams({    {...}    })

iface = stonne.setSubComponent("memory", "memHierarchy.standardInterface")

l1cache = sst.Component("l1cache", "memHierarchy.Cache")
l1cache.addParams({
    "access_latency_cycles" : "1",
    "cache_frequency" : str(tile_clk_mhz) + "GHz",
    "replacement_policy" : "lru",
    "coherence_protocol" : "MESI",
    "associativity" : "16",
    "cache_line_size" : "128",
    "banks" : "32",
    "max_requests_per_cycle" : "64",
    "tag_access_latency_cycles" : "0",
    "verbose" : 10,
    "debug" : 1,
    "debug_level" : 100,
    "L1" : "1",
    "cache_size" : "1024KiB"
})

memctrl = sst.Component("memory", "memHierarchy.MemController")
memctrl.addParams({
    "verbose" : 10,
    "debug" : 1,
    "debug_level" : 100,
    "clock" : str(tile_clk_mhz) + "GHz",
    "addr_range_start" : 0, # full address range
})

backend = memctrl.setSubComponent("backend", "memHierarchy.simpleMem")
backend.addParams({
    "access_time" : "80 ns",
    "mem_size" : str(max_addr_gb) + "GiB"
})

# Enable SST Statistics Outputs for this simulation
sst.setStatisticLoadLevel(statLevel)
sst.enableAllStatisticsForAllComponents({
    "type":"sst.AccumulatorStatistic"
})
sst.setStatisticOutput("sst.statOutputTXT", {
    "filepath" : "output.csv"
})

# Define the simulation links
link_df_cache_link = sst.Link("link_cpu_cache_link")
link_df_cache_link.connect( 
    (iface, "port", "1ps"), 
    (l1cache, "high_network_0", "1ps") 
)
link_mem_bus_link = sst.Link("link_mem_bus_link")
link_mem_bus_link.connect( 
    (l1cache, "low_network_0", "5ps"), 
    (memctrl, "direct_link", "5ps") 
)
```

gvoskuilen, 2024-10-09:
I believe the issue may be that the memHierarchy.standardMem subcomponent is not getting fully initialized because  `m_memInterface->init(phase);` is not called in every phase of `init()`. Try switching the order of that line and the `if (phase != 0) return;` above it in SstMem's `init()` function.

Adrian-2105, 2024-10-10:
Thanks for your answer! If I remember well, I already tried using both ways but I got the same results. But anyway, we ended up doing some workaround and made it work with a previous SST version. I will reopen this issue again if we face this problem once more in the future

closed 2024-10-10 by Adrian-2105