sstsimulator/sst-elements #2346: Modifying firefly parameters in the configuration file

saichennaintel, 2024-04-02:
I'm currently using the network modeling stack of SST (Ember, Firefly, Merlin). Our configuration files use the format mentioned [here ](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/ember/tests/dragon_128_allreduce.py). However we would like to modify the firefly module parameters such as functionSM parameters in configuration file instead of modifying pyfirefly.py and rebuilding it everytime. Can you point out how to update firefly module parameters (such as funcSM, ctrl parameters) in the configuration file? I tried looking into ember and firefly **tests** folder but couldn't find one that would solve the issue.

feldergast, 2024-04-03:
There are two ways to do this.  You can create your own platform file and load it instead of the default one in pyember.  You can also overwrite any of the defaults from the platform file in the python input file.  I'll pull together an example of each and post it later today.

feldergast, 2024-04-03:
For defining firefly parameters in your input file you would set them in the EmberMPIClass.  All the Firefly functions are found under either the "nic" or "os" variable. As an example:

ep = EmberMPIJob(...)
ep.nic.nic2host_lat = "150ns"
ep.os.ctrl.shortMsgLength = 15000
ep.os.functionsm.smallCollectiveSize = 64

You can look at pyfirefly.py to find which options fall under which section.subsection (i.e. nic, os.functionsm, os.ctrl, etc).  In that file, here are the mappings based on where you find the option.

BasicNicConfiguration : nic
FireflyHades: os
  There are two more levels here: functionsm and ctrl

Hopefully, that will allow you to map all the options to their proper full name in the Python.

This can be done on a per job basis.  If you want all jobs to have the same parameters, then I would suggest deriving a class from EmberMPIJob and setting the variables in the constructor (replace "ep." with "self." from the above example), then using that class to create jobs.

saichennaintel, 2024-04-05:
Thank you @feldergast ! This is really useful.

closed 2024-04-05 by saichennaintel