sstsimulator/sst-elements #2358: Running Vanadis hello-world with testScratchCache-3.py hierarchy

plafratt, 2024-05-16:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

I modified vanadis/tests/basic_vanadis.py so that the L1, L2, and L3 are configured like the L1, L2, and L3 in `memHierarchy/tests/testScratchCace-3.py`. When I run the `hello-world` program, it hangs.

When I change the L2 and L3 from noninclusive to inclusive, the simulation runs to completion successfully.

Is this configuration supported with noninclusive caches?

I also tried using the L1, L2, and L3 configuration in `memHierarchy/tests/testNoninclusive-1.py`, and that also hung.

2 - Describe how to reproduce

Add an L3 to `vanadis/tests/basic_vanadis.py`. Then configure the L1, L2, and L3 to match the L1, L2, and L3 in `memHierarchy/tests/testScratchCace-3.py` or `memHierarchy/tests/testNoninclusive-1.py`. Then run `hello-world` with `basic_vanadis.py`.

I've attached my modified version of `basic_vanadis.py` with these changes. [basic_vanadis_noninclusive.tar.gz](https://github.com/sstsimulator/sst-elements/files/15339031/basic_vanadis_noninclusive.tar.gz)

3 - What Operating system(s) and versions 

Ubuntu 20.04.

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

sst-core - d66f07e4b
sst-elements - 2f81e3a1fdf7

6 - Fill out Labels, Milestones, and Assignee fields as best possible

plafratt, 2024-08-27:
I was looking at this today and noticed an issue with the example I'd posted above. I fixed it in this version. 
[basic_vanadis_noninclusive_2024-08-27.tar.gz](https://github.com/user-attachments/files/16765403/basic_vanadis_noninclusive_2024-08-27.tar.gz)

hughes-c, 2024-12-18:
I believe that https://github.com/sstsimulator/sst-elements/pull/2430 has some bug fixes for non-inclusive caches.

plafratt, 2024-12-18:
Thank you for the update!