sstsimulator/sst-elements #2361: Ariel Core CPU cannot connect to Shared Type elements

dmukherj09, 2024-05-27:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

2 - Describe how to reproduce

3 - What Operating system(s) and versions 

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

gvoskuilen, 2024-07-03:
This is side-effect of memHierarchy using component names to identify event senders. To fix, you'll need to use a shim that ensures L1 requests/responses keep their core IDs since they're all coming from a single ariel component and getting sent to the same cache. The shim is called "memHierarchy.multithreadL1" and can be used instead of the bus between the cores & the cache. Something along these lines:
```py
ariel = sst.Component(...)
cache = sst.Component(...)
shim = sst.Component("cores_to_cache", "memHierarchy.multithreadL1")
shim.addParams( ... ) # Set clock and number of requests/responses that can be handled each cycle
for x in range(0, cores):
    link = sst.Link("link_core_" + str(x))
    link.connect( (ariel, "cache_link_" + str(x), latency), (shim, "thread" + str(x), latency) )
cachelink = sst.Link("link_cache")
cachelink.connect( (shim, "cache", latency), (cache, "high_network_0", latency) )
```
If it's helpful, the memHierarchy test "tests/testKingsley.py" uses the multithreadL1 to connect two cores per first level cache.

jwilso, 2024-09-19:
Closing as resolved.

closed 2024-09-19 by jwilso