sstsimulator/sst-elements #2365: How do I let Merlin know what's the dest of outgoing Event?

cainingw, 2024-06-06:
Taking the simpleElementExamlpe as an example, I created multi simpleElementExamlpe components, and I connect each of them to a Merlin router, when I use link->send(event) in the simpleElementExamlpe component to send an Event to another simpleElementExamlpe component, how do I let Merlin router know which component this Event is going to, or how I set dest for this Event?

feldergast, 2024-06-10:
The SimpleElementExample components don't have the right SubComponent interface to connect up to merlin routers.  There are two parts to the merlin library, the routers and the LinkControl, which inherits from the SimpleNetwork interface in the core (set/core/interfaces/simpleNetwork.h).  A better example for connecting to merlin would be either sst/elements/merlin/test/nic.{h,cc}, or sst/elements/memHierarchy/memNic.{h,cc}.  An example of how to configure a merlin simulation can be found in /sst/elements/merlin/tests/dragon_128_platform_test.py.

At the lowest level, the SimpleNetwork interface uses the SimpleNetork:Request object to specify the destination of the packet.  The destination corresponds to the ID returned by the SimpleNetwork::getEndpointID() function for each of the endpoint components.  The link connecting the endpoint to the network is actually connected to the LinkControl object and sends and receives are done through the SimpleNetwork API.  The nic.cc example listed above should give you a good idea of how all this works.

cainingw, 2024-06-11:
> The SimpleElementExample components don't have the right SubComponent interface to connect up to merlin routers. There are two parts to the merlin library, the routers and the LinkControl, which [...]

Thank you so much!

closed 2024-06-11 by cainingw