sstsimulator/sst-elements #2368: balar build problem, no libcudart_mod.so

cainingw, 2024-06-11:
I followed build instruction at https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/balar/README.md

After these steps:

# Setup GPGPU-Sim in SST-integration mode
source setup_environment sst

# Build GPGPU-Sim
make -j

I foud there are only files named libcudart.so in \gpgpu-sim_distribution\lib\gcc-7.5.0\cuda-10010\release

But balar needs libcudart_mod.so to build.

jwilso, 2024-09-19:
@hughes-c Is this a CUDA issue?

William-An, 2024-10-02:
You probably will need to checkout to `sst-integration` branch for the GPGPU-Sim repo. We are working on a PR accel-sim/gpgpu-sim_distribution#44 into our GPGPU-Sim, which should get rid of this extra step.

closed 2024-10-24 by hughes-c