sstsimulator/sst-elements #2404: Balar does not compile as written

mgoldstein322, 2024-10-02:
Compiling balar requires the user to comment out line 19 of `balarTestCPU.cc`. This line is `#include <sst/core/simulation.h>`, which does not exist. Replacing the line with `#include <sst/core/simulation_impl.h>` does not solve the compilation issue. Additionally, running the balar tests afterward produces a segmentation fault.

How to reproduce:

Compile balar by following the instructions at https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/balar/README.md but comment out line 19 of `balarTestCPU.cc` so that compilation will complete. Next, run `sst testBalar-testcpu.py --model-options='-c gpu-v100-mem.cfg -v -x vectorAdd/vectorAdd -t vectorAdd/cuda_calls.trace'`. The test will complete and display the "Test Completed Successfully" message, then seg fault.

RHEL9 -- CUDA 11 -- SST core commit 0075ba17 -- SST elements commit 4c91414f

jwilso, 2024-10-14:
@hughes-c Any progress on this issue? It would be nice to get it into 14.1.0.

jwilso, 2024-10-14:
@gvoskuilen This will be deferred to a later release. For 14.1.0, documentation will note this as an open issue.

hughes-c, 2024-12-18:
Closing this since https://github.com/sstsimulator/sst-elements/pull/2428 makes changes to balar and the build requirements.

closed 2024-12-18 by hughes-c