sstsimulator/sst-elements #2409: A building problem in ember

chaojibendan, 2024-10-16:
When try to use `--enable-ember-contexts` in configure, I encouter a building problem. It seems including `header file path error` and `compiler tool version error`. Such as `#include "embermpigen.h"` in `embermpicxtgen.h`, it should be `mpi/embermpigen.h`. As for `compiler tool version error`, I use gcc 11.4.0.

feldergast, 2024-10-16:
The code in question looks like a stub for a future feature.  It was last touched about 10 years ago, so I think it's safe to assume the feature was abandoned. We will remove the configure option and code to avoid any confusion in the future.

berquist, 2026-05-13:
This was removed in https://github.com/sstsimulator/sst-elements/pull/2684.