sstsimulator/sst-elements #2491: distclean Inadvertently Removes Files From In-Source Builds

hughes-c, 2025-04-03:
When not doing out-of-source builds, distclean removes these files from the source directory:
        src/sst/elements/ariel/api/arielapi.c
        src/sst/elements/ariel/api/arielapi.h
        src/sst/elements/ariel/mpi/fakepin.cc
        src/sst/elements/ariel/mpi/mpilauncher.cc

https://github.com/sstsimulator/sst-elements/commit/6ecf9bde3350699455b27105daaf81550cf6c813

hughes-c, 2025-04-04:
@plavin Is there a reason these files need to be copied into build? They can't be referenced from source?

plavin, 2025-04-04:
Thanks for fixing that. I had a WIP fix but hadn't pushed it yet. I had issues with out-of-source builds that I was attempting to workaround. Referencing the source is a better solution.

closed 2025-04-15 by hughes-c