sstsimulator/sst-elements #2520: Failed to Run Ariel CPU Model Using SST-Elements Built with PinTool

chenshih1, 2025-05-07:
I built SST-Elements 14.1.0 with Pin 3.30 following the instructions provided on the official websites, and the compilation completed successfully. The pinfrontend component of Ariel is able to locate and invoke the pinbin executable correctly. 

However, despite adding the path to the PinTools shared libraries to environment variables such as LD_LIBRARY_PATH, I encounter errors when running the example code from the [sst-tutorials](https://github.com/sstsimulator/sst-tutorials) repository — specifically when using Ariel CPU configurations.

When attempting to run ex4 from the sst-tutorials using Ariel CPU configurations, I encounter the following error.

The error reported by SST suggests a failure related to loading or initializing the Pin-based frontend, likely due to a missing or misconfigured shared library dependency, although all required paths appear to be set correctly. If I run pinbin outside of SST-Elements, it executes smoothly without any issues.

```
pin/intel64/bin/pinbin: error while loading shared libraries: libxed.so: cannot open shared object file: No such file or directory
FATAL: Pin3Frontend[frontend/pin3/pin3frontend.cc:444:forkPINChild] Launching trace child failed!  Child Exited with status 127
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3f6) [0x55e6af68b176]
    1 : /home/chens/local/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent12Pin3Frontend12forkPINChildEPKcPPcRSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_St4lessISC_ESaISt4pairIKSC_SC_EEENS0_15redirect_info_tE+0x7bd) [0x7f2d44a263cd]
    2 : /home/chens/local/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent12Pin3Frontend4initEj+0x136) [0x7f2d44a265e6]
    3 : /home/chens/local/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4initEj+0x21) [0x7f2d449e08c1]
    4 : sst(_ZN3SST15Simulation_impl10initializeEv+0x94) [0x55e6af6b0a24]
    5 : sst(+0x135991) [0x55e6af5f7991]
    6 : sst(main+0x640) [0x55e6af5d76f0]
    7 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f2d4d1ae083]
    8 : sst(_start+0x2e) [0x55e6af5f686e]
```

Operating System: Ubuntu 20.04.6 LTS

CPU: AMD Ryzen 7 3700X 8-Core Processor

Compiler Version: GCC 9.4.0

chenshih1, 2025-05-07:
The following paths, which point to all the necessary shared libraries required by PinTool, have been set via LD_LIBRARY_PATH:
```
/usr/local/pin/intel64/runtime/pincrt
/usr/local/pin/intel64/lib
/usr/local/pin/intel64/lib-ext
/usr/local/pin/extras/xed-intel64/lib
```
However, if I instead add the same paths to a configuration file (pin.conf) under /etc/ld.so.conf.d/, the error changes.

Here is the new error message:
```
E: Unable to load /usr/local/libexec/fesimple.so
FATAL: Pin3Frontend[frontend/pin3/pin3frontend.cc:373:forkPINChild] Launching trace child failed!  Child Exited with status 255
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x4ae) [0x5567706ae67e]
    1 : /usr/local/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent12Pin3Frontend12forkPINChildEPKcPPcRSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_St4lessISC_ESaISt4pairIKSC_SC_EEENS0_15redirect_info_tE+0x7d2) [0x7fd298ef0572]
    2 : /usr/local/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent12Pin3Frontend4initEj+0x147) [0x7fd298ef0717]
    3 : /usr/local/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4initEj+0x21) [0x7fd298ea5561]
    4 : sst(_ZN3SST15Simulation_impl10initializeEv+0xa4) [0x5567706d77b4]
    5 : sst(+0xf9168) [0x55677061d168]
    6 : sst(main+0x188b) [0x556770602f0b]
    7 : /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fd29c9afd90]
    8 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fd29c9afe40]
    9 : sst(_start+0x25) [0x55677061a735]
```
What is the cause of this issue? How can I resolve it?

gvoskuilen, 2025-05-07:
This means Ariel forked the pin process to trace the simulated executable but the pin process exited unexpectedly. ex4 uses a separate config file (ariel.cfg) to parameterize the SST input file. Assuming you are using the example input files, does the executable listed in that file (`../stream`) exist?

chenshih1, 2025-05-08:
> This means Ariel forked the pin process to trace the simulated executable but the pin process exited unexpectedly. ex4 uses a separate config file (ariel.cfg) to parameterize the SST input file. [...]

Yes, before entering the directory to execute ex4.py, I had already run the make command at the parent level, which successfully generated the stream executable file. The executable should exist and be correctly referenced in the configuration file.

plavin, 2025-05-16:
It looks like we are using some outdated configuration options in that example. It is setting an `application` parameter that no longer exists in the Ariel CPU, and it is not properly setting the executable. 

Here is a temporary fix for `ex4.py` until we can fix the sst-tutorials repo:

```diff
 if 'ariel' in config.app:
     arielCPU = sst.Component("A0", "ariel.ariel")
-    arielCPU.addParams(config.getCoreConfig(0))
+    arielCPU.addParams({'clock':'2660MHz',
+                        'num_cores': 8,
+                        'max_reqs_cycle': 3,
+                        'executable' : '../stream'})
```

chenshih1, 2025-05-19:
> It looks like we are using some outdated configuration options in that example.
> 
> Here is a temporary fix for `ex4.py` until we can fix the sst-tutorials repo:
> 
>  if 'ariel' in config.app:
>      arielCPU = sst.Component("A0", "ariel.ariel")
> -    arielCPU.addParams(config.getCoreConfig(0))
> +    arielCPU.addParams({'clock':'2660MHz',
> +                        'num_cores': 8,
> +                        'max_reqs_cycle': 3,
> +                        'executable' : '../stream'})
> You can also find more Ariel examples hidden in `sst-elements/src/sst/elements/ariel/frontend/simple/examples/stream`.

Thanks for the suggestion! After applying the recommended changes to the configuration, all the examples in the sst-tutorials/exercises directory are now able to run successfully using ariel.cfg. I'll study the examples in the SST elements source code to learn how to use the SST simulator properly. Thank you once again for your help!

closed 2025-05-22 by gvoskuilen