sstsimulator/sst-elements #2560: [Question] Why are prefetches dropped?

jake-ke, 2025-08-15:
I have set the L1 cache controller (no other caches) and the memory controller to have infinite processing capability, such as:
- `max_requests_per_cycle`
- `request_link_width/response_link_width`
- `mshr_num_entries`
- `request_width`

However, the`Prefetch_drops` stats shows significant (~50%) drops.

gvoskuilen, 2025-08-27:
There are a number of reasons that the cache drops prefetches. Reasons include (and some of these are unlikely given how you've configured your L1):
1) There is already an outstanding request for the prefetch address or there was a request for that address that hit in the same cycle the prefetch was handled.
2) In the cycle when the cache attempted to handle a prefetch, the `max_requests_per_cycle` was already met
3) `max_outstanding_prefetch` has been met - this is the maximum number of pending prefetches allowed at a time and defaults to half the size of the MSHR
4) `drop_prefetch_mshr_level` - prefetches are dropped if the total MSHR occupancy (including regular requests and prefetches) exceeds this.

(2) is unlikely given your configuration, but the others may be occurring. If you want to generate a trace of cache and prefetch accesses to see what is happening, you can do so - you would need to configure sst-core with `--enable-debug` and then set the cache debug parameters (`debug=1` and `debug_level >= 5`).

jake-ke, 2025-08-28:
Thank you for the answers! 
Does (1) also include prefetches that are late (requests for the same address shortly issued after prefetch)?